efabless/caravel_mgmt_soc_litex #1: Make sure design is able to boot on a Digilent Arty A7 board

mithro, 2021-10-30:
The Digilent Arty A7 board is an reasonable cheap, fairly large, modern and extremely common FPGA development board.

jeffdi, 2022-09-27:
Currently supported on CMOD7.  Will share repo publically.

proppy, 2022-12-13:
@jeffdi is there any public documentation on how to get the design working on a CMOD/A7 board?

jeffdi, 2022-12-15:
> @proppy <https://github.com/proppy> @jeffdi <https://github.com/jeffdi>
> we have a WIP branch
> <https://github.com/antmicro/caravel-lite/commits/arty> from which you
> can build a bitstream for Arty A7. Simply run make install_mcw to get an updated
> caravel_mgmt_soc_litex
> <https://github.com/antmicro/caravel_mgmt_soc_litex/tree/litex-lock> and
> then make vivado to build the bitstream.
> For the CMOD A7 it should be just a matter of replacing the XDC file.

Thanks for this.  We will have a look.