efabless/caravel_mgmt_soc_litex #11: mgmt_core/user_irq input_delay missing?

mbalestrini, 2022-01-04:
I'm running STA on the whole caravel+user_project_area to try to detect any issues we might get from the integreation of both. 
I have a question about the mgmt_core.user_irq inputs:

In the new core there seems to be a lot of hold delays cells to handle outputs from the user_project_wrapper (`la_data_out` , `wbs_dat_o`) into the mgmt_core that help even out the difference in clocks to avoid hold violations. But it seems there are no delay cells on the `user_irq` paths, and that is causing some hold violations in our design.
Taking a quick look at the `mgmt_core.sdc` I see that it sets an `input_delay` of 1.0 to all the inputs (`mprj_dat_i`, `la_input`)  but not for the `user_irqs`. Is there a reason behind that?  I saw that the user_irq inputs use double flip-flop to sync the signal but when the signal comes from a synced cell in the projet area it might probably always caused a hold violation on the first ff

mbalestrini, 2022-01-04:
Here's a thread on the slack where I posted about this topic: 
https://skywater-pdk.slack.com/archives/C02K4RD241Y/p1640276417199900

azwefabless, 2022-10-14:
Addressed with  a preliminary input constraint. Not clear that the general problem is addressed with clear boundary constraints.