efabless/caravel_mgmt_soc_litex #21: mprj wishbone address range is limited to 0x3010_0000 instead of 0x3fff_ffff

mattvenn, 2022-03-03:
https://github.com/efabless/caravel_mgmt_soc_litex/blob/3fee299f8177cebf7919eb0e6da7f0f5ad7af31d/litex/caravel.py#L186

mattvenn, 2022-03-11:
@suppamax said: 

> (Here)[https://caravel-harness.readthedocs.io/en/latest/memory-mapped-io-summary.html] a wishbone base address is mentioned: should I add a "wishbone last addr" field?
> 
> I cannot find a wishbone-related documentation in the management SoC repo (and it looks like there is no readthedocs page yet).

Those docs are out of date from when caravel contained the cpu as well. Now this repo contains the litex management soc containing the cpu so this repo should generate the docs. You can see them in https://github.com/efabless/caravel_mgmt_soc_litex/tree/main/docs/generated

so far they don't have a readthedocs (also needs to get done).

RE your question, I'd say yes, either an end address or a width would work.

suppamax, 2022-03-11:
Do I understand it correctly - the idea is to 
1. extend that outdated document with the warning (size or end addr)
2. consolidate the documentation for `caravel_mgmt_soc_litex`

mattvenn, 2022-03-11:
I think leave the docs stuff for me. 
The idea is we will have at least 2 repos providing docs for caravel:
* caravel which contains, gpios, the management protection, spi stuff
* litex (this repo) that contains everything inside the managment soc, which would include wishbone, memory maps. etc.

Make sense?

egorxe, 2022-06-03:
May be it would be better to extend this address space for future MPWs in addition to documenting it? 1 MByte may be not enough for some projects (SDRAM/HyperRAM controllers for example) and there seems to be no reason why it could not be extended up to previous size (256 MBytes). Currently this address space is empty anyway.

clp510, 2022-07-14:
is there any update on this bug ?

jeffdi, 2022-09-26:
duplicates issue #10

closed 2022-09-26 by jeffdi