efabless/caravel_mgmt_soc_litex #32: add instructions to README.md on how to run tests

mattvenn, 2022-03-23:
tests are confusing, what is the difference between tests-standalone and tests-caravel
I'm unable to run any tests
the run.sh and run-all.sh scripts fail
running make from the individual test directories fail, how to run these tests?

mattvenn, 2022-03-23:
@suppamax - can you comment on what steps you took to run the new irq test? I want to approve your PR but I can't even run the test!

suppamax, 2022-03-23:
I think the only necessary step is to define the `DESIGNS` variable.

I have the following structure
```
├── caravel
    ├── caravel
    ├── caravel_mgmt_soc_litex
    └── caravel_user_project
```
and for me `DESIGNS = /path/to/caravel`

Hope this helps.

DavidRLindley, 2024-12-10:
I'm closing due to the age and staleness.

closed 2024-12-10 by DavidRLindley