efabless/caravel_mgmt_soc_litex #51: SRAM second port has "no connect" inputs

RTimothyEdwards, 2022-10-04:
caravel_mgmt_soc_litex verilog module `verilog/rtl/mgmt_core_wrapper.v` has the pins `sram_ro_clk`, `sram_ro_csb`, and `sram_ro_addr` are no-connects.  These *must* be connected, to ground if nothing else.  Also see the issue in the caravel repository issue number 6---this value could be controlled by housekeeping through the SPI, which is the preferable implementation if it has to be fixed anyway.

jeffdi, 2022-10-06:
RTL implemented.  Just needs a testbench.

RTimothyEdwards, 2022-10-07:
This one will not be fixed since we have made the decision to remove the SRAM and replace it with DFFRAM (which doesn't have a 2nd read-only port).  The read-only ports have been removed from the LiteX core and `ifdef'd out of the housekeeping and top levels.