efabless/caravel_mgmt_soc_litex #52: Different definitions  at caravel.h and defs.h

M0stafaRady, 2022-10-05:
[Caravel.h](https://github.com/efabless/caravel_mgmt_soc_litex/blob/caravel_redesign/verilog/dv/firmware/caravel.h) and [defs.h](https://github.com/efabless/caravel_mgmt_soc_litex/blob/caravel_redesign/verilog/dv/firmware/defs.h) have different declaration for  `GPIO_MODE*` definitions. I tried the `GPIO_MODE_MGMT_STD_INPUT_PULLDOWN `and `GPIO_MODE_MGMT_STD_INPUT_PULLUP `in both and the one in caravel.h seems to be correct as the other one don't even configure them as inputs. 
Also register `reg_wb_enable `is declared 2 times with different values.

M0stafaRady, 2022-10-05:
Also configuring the gpio as pull up or pull down don't seem to work I can see X's over housekeeping.mgmt_gpio_in signal when not diving

RTimothyEdwards, 2022-10-07:
This has been fixed (needs to reference a pull request)

closed 2022-10-08 by M0stafaRady

jeffdi, 2022-11-01:
Fixed in ae92b99c17c4014daeff3021656ec19fecacb1e6