efabless/caravel_mgmt_soc_litex #87: add pass through signals to the `mgmt_core_wrapper`

passant5, 2022-10-14:
The following signals are required to pass through the `mgmt_core_wrapper`:
1. `housekeeping/serial_load`
2. `housekeeping/serial_data_2`
3. `housekeeping/serial_resetn`
4. `housekeeping/serial_clock`
5. `xres_buf/rstb_l`
6. `simple_por/por_l`
7. `simple_por/porb_h`

To do this, we need 7 input ports `<signal_name>_in` and 7 output ports `<signal_name>_out`

jeffdi, 2022-10-14:
@passant5 @shalan signal name for housekeeping/serial_load would be serial_load_in and serial_load_out ?

passant5, 2022-10-14:
@jeffdi yes

M0stafaRady, 2022-10-14:
@jeffdi  your last [commit ](https://github.com/efabless/caravel_mgmt_soc_litex/commit/1321c51aeedc19405f5a20b68edc5bdc4fab5c57) added these signals to the mgmt_core only but not the mgmt_core_wrapper module. That broke the RTL simulation because of @passant5  [last commit](https://github.com/efabless/caravel/commit/653e7fa56142f6b2c8faf2c069354b607a88668c) at caravel repo.

RTimothyEdwards, 2022-10-20:
FYI, before closing this ticket, I want to mention that #7 is a 3.3V signal and should not have been on the list.  This error was corrected later.

closed 2022-10-20 by RTimothyEdwards